sav-oss-eplanner/EasyEPLANner #1182: Копирование операции с привязанными узлами

KUZMA-1989, 2023-06-14:
Сейчас, когда копируешь операцию с привязанными узлами и параметрами, то когда вставляешь в том же объекте, то всё сбрасывается и привязка узлов и параметры, нужно чтоб оставалось

KirillGutyrchik, 2023-06-15:
@KUZMA-1989, необходимо полностью сбрасывать всю привязку для объекта или только доп.свойства агрегатов для данной опирации?

KUZMA-1989, 2023-06-15:
> @KUZMA-1989, необходимо полностью сбрасывать всю привязку для объекта или только доп.свойства агрегатов для данной опирации?

Только в той операции сброс доп свойств, которую меняешь

KUZMA-1989, 2023-06-15:
> @KUZMA-1989, необходимо полностью сбрасывать всю привязку для объекта или только доп.свойства агрегатов для данной опирации?

Только доп свойства

commit 345b07fc (2023-08-14): Dialog reset extra properties attached objects (#1192)

Fixes #1182.
PR #1192 (2023-08-14): Dialog reset extra properties attached objects
Fixes #1182.
closed 2023-08-14 by idzm