sav-oss-eplanner/EasyEPLANner #1198: Переименовать подписи к функционалу шагов

KUZMA-1989, 2023-08-02:
[image]

KUZMA-1989, 2023-08-02:
@KirillGutyrchik Добавь в этот pullrequest и вот это 
[image]

idzm, 2023-08-04:
Для большей информативности предлагаю использовать такие названия:
`Сигнал для включения шага` -> `Сигнал для включения данного шага`;
`Перейти в шаг по условию` -> `Переход к другому шагу по условию`.

@KUZMA-1989, @AndreyNemenkov - что скажете?

KUZMA-1989, 2023-08-04:
> Для большей информативности предлагаю использовать такие названия: `Сигнал для включения шага` -> `Сигнал для включения данного шага`; `Перейти в шаг по условию` -> `Переход к другому шагу по условию`.
> 
> @KUZMA-1989, @AndreyNemenkov - что скажете?

Можно так

KUZMA-1989, 2023-08-04:
> Для большей информативности предлагаю использовать такие названия: `Сигнал для включения шага` -> `Сигнал для включения данного шага`; `Перейти в шаг по условию` -> `Переход к другому шагу по условию`.
> 
> @KUZMA-1989, @AndreyNemenkov - что скажете?

Обсудили в отделе, оставляем мои подписи : Сигнал для включения текущего шага, Переход к следующему шагу по условию

PR #1199 (2023-08-04): Rename display names actions jump_if and enable_step_by_signal
Fixes #1198.
commit cb88cedc (2023-08-04): Rename display names actions jump_if and enable_step_by_signal (#1199)

Fixes #1198.
closed 2023-08-04 by idzm