sav-oss-eplanner/EasyEPLANner #1204: Ошибка предельного регулятора во время теста

AndreyNemenkov, 2023-08-25:
"В выходе OUT_VALUE ПИД-регулятора ZM51LTC1 задано некорректное устройство. Нужно указать AO, VC, M или другой ПИД-регулятор." Появляется это сообщение при добавлении предельного регулятора
, в выход прописывал клапан.

KirillGutyrchik, 2023-08-25:
@AndreyNemenkov, по документации [6.5 Работа с ПИД-регулятором](https://github.com/savushkin-r-d/EasyEPLANner/blob/master/docs/user_manual/ReadMe.md#65-работа-с-пид-регулятором) для свойства OUT_VALUE могут быть заданы только устройства с типами: AO, VC, M или C(ПИД-регулятор). Нужно добавить возможность так же подключения типа V?

idzm, 2023-09-04:
Имеется в виду `C_THLD` - пороговый регулятор. Для него выход является дискретным устройством (любой клапан, насос).

KirillGutyrchik, 2023-09-04:
@idzm, @AndreyNemenkov, помимо **DO**, **V** и **M**, какие еще устройства могут подключаться к выходу порогового регулятора, а также доступно ли каскадное подключение нескольких пороговых регуляторов в цепочку?

idzm, 2023-09-04:
Каскадного регулирования не будет. По устройствам - здесь все те устройства, которые возможны для действия `"Включение"`.

commit 95207e5e (2023-09-06): Fix allowed devices of C_THLD (#1209)

Fixes #1204.
PR #1209 (2023-09-06): Fix allowed devices of C_THLD
Fixes #1204.
closed 2023-09-06 by idzm