sav-oss-eplanner/EasyEPLANner #1221: Сброс привязанных параметров и объектов

KUZMA-1989, 2023-10-16:
Заменил в линии наполнения операцию "работа" на "наполнение", выскачило сообщение "сбросить доп свойства привязанных аггрегатов", нажад "Нет", а программа все равно всё сбросило
[image]
[image]

KirillGutyrchik, 2023-10-17:
@KUZMA-1989, у вас в объекте нет привязанных агрегатов, данная функция работает только с их параметрами. Базовые шаги сбрасываются всегда по умолчанию, если их нужно также не сбрасывать, могу это добавить.

KUZMA-1989, 2023-10-17:
> @KUZMA-1989, у вас в объекте нет привязанных агрегатов, данная функция работает только с их параметрами. Базовые шаги сбрасываются всегда по умолчанию, если их нужно также не сбрасывать, могу это добавить.

Да, добавь

commit 7f43fc22 (2023-10-17): Don't reset base steps after dialog "reset extra properties" (#1222)

Fixes #1221.
PR #1222 (2023-10-17): Don't reset base steps after dialog "reset extra properties"
Fixes #1221.
closed 2023-10-17 by idzm