sav-oss-eplanner/EasyEPLANner #1290: Орфографическая ошибка

KapuzaVladAC-62, 2023-12-14:
в 26 строке readme вместо " operations on these objects" указано " operations of these objects"
ссылка на файл:https://github.com/KapuzaVladAC-62/EasyEPLANner/blob/master/ReadMe.md

KirillGutyrchik, 2023-12-14:
@KapuzaVladAC-62, у нас описываются операции объектов, а не операции выполняемые над объектами, _*of*_ - правильный вариант, или я ошибаюсь?

KapuzaVladAC-62, 2023-12-14:
@KirillGutyrchik правильный вариант on. В английском языке предлог "on" в этом контексте используется для того, чтобы указать, что действие (operations) направлено на определенные объекты (technological objects, такие как Tank, Boiler и т.д.). Это связано со спецификой функционирования объекта либо действиями, которые осуществляются непосредственно в отношении этого объекта.

Предложение "operations on these objects" означает, что пользователь описывает операции, применяемые к этим технологическим объектам, а не где-то еще. Предлог "on" в данном случае подчеркивает фокус на воздействии или обработке этих конкретных объектов.

Вот некоторые примеры использования подобной структуры в английском языке:

Performing maintenance on the machinery. (Проведение техобслуживания на машинах.)
Conducting experiments on the samples. (Проведение экспериментов на образцах.)
Executing commands on the server. (Выполнение команд на сервере.)
Во всех случаях предлог "on" связывает действие с объектом, на который это действие оказывает влияние.

KirillGutyrchik, 2023-12-15:
@KapuzaVladAC-62, у нас согласно стандарту, технологический объект это группа оборудования более низкого уровня. И операции выполняются этими тех. объектами управляя оборудованием. Для оборудования да, можно применить _*on*_. В данном случае наш объект является как-бы источником или исполнителем операции, а не таргетом, на который эта операция окажет влияние.

closed 2023-12-20 by idzm