sav-oss-eplanner/EasyEPLANner #1339: Добавить новый тип клапан As-interface

KUZMA-1989, 2024-03-13:
Добавить новый тип клапана с As-interface **V_AS_MIXPROOF_V70**. Описание аналогично **V_AS_MIXPROOF**

rartem, 2024-03-13:
[thinktop-v50-and-v70-product-leaflet.pdf](https://github.com/savushkin-r-d/EasyEPLANner/files/14585590/thinktop-v50-and-v70-product-leaflet.pdf)
[think top.PDF](https://github.com/savushkin-r-d/EasyEPLANner/files/14585593/think.top.PDF)
Раньше у них походу не было подписано, где верхнее и где нижнее.

KUZMA-1989, 2024-03-13:
> [thinktop-v50-and-v70-product-leaflet.pdf](https://github.com/savushkin-r-d/EasyEPLANner/files/14585590/thinktop-v50-and-v70-product-leaflet.pdf) [think [...]

@rartem Для старых типов клапанов AS-i оставляем по старому описание, для новых нужно поменять команды для флипов седел

idzm, 2024-03-14:
В EPLANner изменения не требуются - только в управляющей программе.

closed 2024-03-27 by KirillGutyrchik