sav-oss-eplanner/EasyEPLANner #1350: Сигнальная колонна

SergeiSilakov, 2024-03-22:
При использовании дискретной сигнальной колонны параметр в документации R_CONST_RED указан не верно при "1" - если мигающий, при "0" если постоянный
https://github.com/savushkin-r-d/EasyEPLANner/blob/master/docs/user_manual/ReadMe.md#:~:text=HLA-,R_CONST_RED,-%2D%20%D0%A2%D0%B8%D0%BF%20%D0%BA%D1%80%D0%B0%D1%81%D0%BD%D0%BE%D0%B3%D0%BE%20%D1%81%D0%B8%D0%B3%D0%BD%D0%B0%D0%BB%D0%B0

idzm, 2024-03-26:
> При использовании дискретной сигнальной колонны параметр в документации R_CONST_RED указан не верно при "1" - если мигающий, при "0" если постоянный [...]

А что это за ссылка такая? Куда она указывает?

SergeiSilakov, 2024-03-26:
Должна была на параметр R_CONST_RED

idzm, 2024-03-26:
Надо исправить. Вот пример корректной ссылки:
https://github.com/savushkin-r-d/EasyEPLANner/blob/8090cd4873f0002cfc2b836e26c3bf6e959a6064/docs/user_manual/ReadMe.md?plain=1#L1123-L1124

PR #1351 (2024-03-26): [manual] Fix HLA R_CONST_RED value description
Fixes #1350.
commit 381b5ca7 (2024-03-26): [manual] Fix HLA R_CONST_RED value description (#1351)

Fixes #1350.
closed 2024-03-26 by idzm