sav-oss-eplanner/EasyEPLANner #1361: Вывод сообщения об ошибке привязки клапанов с некорректным типом

KUZMA-1989, 2024-04-17:
Сейчас есть возможность привязать клапана типа
**V_DO1
V_DO2
V_DO1_DI1_FB_OFF
V_DO1_DI1_FB_ON
V_DO1_DI2
V_DO2_DI2**
к пневмооcтрову IO_LINK 
**DEV_VTUG_8
DEV_VTUG_16 
DEV_VTUG_24**
Нужно сделать, чтоб не было возможности привязки. Или если они привязались, чтоб выдавалось сообщение о данном событии в логах проекта

KirillGutyrchik, 2024-04-24:
@KUZMA-1989, для уточнения:
1. Для DI каналов подтипов указанных в первой части оставить возможность привязки к пневмоострову?
2. По iolink клапанам это касается только DO модулей или также DI/DO?

KUZMA-1989, 2024-04-24:
> @KUZMA-1989, для уточнения:
> 1. Для DI каналов подтипов указанных в первой части оставить возможность привязки к пневмоострову?
> 2. По iolink клапанам это касается только DO модулей или также DI/DO?

Di к пневмоострову привязывать нельзя

PR #1368 (2024-05-02): Add check binding valve subtypes to IO-module/valve terminal
Fixes #1361.

```ChangeLog
Добавлена проверка привязки клапанов некоторых подтипов к пневмоострову и модулю ввода вывода
```
commit 0541aab9 (2024-05-02): Add check binding valve subtypes to IO-module/valve terminal (#1368)

Fixes #1361.

```ChangeLog
Добавлена проверка привязки клапанов некоторых подтипов к пневмоострову и модулю ввода вывода
```
closed 2024-05-02 by idzm