sav-oss-eplanner/EasyEPLANner #1399: Удаление устройств при изменении номера объекта

SergeiSilakov, 2024-07-09:
[image]
При изменении номера объекта епланер удаляет устройства из описания операций с номером объекта на который был изменён.
Например был №1, меняю на №12 - Устройства описанные в операциях с номером ОУ №12(Х12V1) будут удалены.

KirillGutyrchik, 2024-07-15:
@SergeiSilakov, удалять устройства не должно, вместо этого меняется номер объекта в устройствах на: Q21V1 -> Q11V1. Если такого клапана нет, то да, устройство будет удалено.

SergeiSilakov, 2024-07-15:
@KirillGutyrchik ну у меня же получилось, у меня есть устройства и они уже были прописаны с ОУ Q21Vx
Их удалило когда я менял номер с 11 на 21 на скрине это видно

KirillGutyrchik, 2024-07-15:
@SergeiSilakov, ну да, я про это и пишу, когда номер устройства изменяется: 11 -> 21, а до этого были привязаны устройства объекта 21, то вместо того, чтобы оставлять эти устройства как есть, они изменяются на старый номер. То есть, например если в устройсве были прописаны устройства `Q11V1` `Q21V2`, то после изменения номера получится: `Q21V1` `Q11V2` (а так как в данном случае Q11V2 видимо отсутствует на схеме, то оно удаляется из привязки).
Это всегда так работало. Я могу попробовать это поправить, но то, что это не вызовет ошибок в других местах после этого, я гарантировать не могу.

commit 8f5df4ad (2024-07-17): Fix modify dev obj name (#1400)

Fixes #1399.

```ChangeLog
Изменения в модификации названий устройств: если при изменении номера объекта (1 -> 2) в объекте было привязаны устройства OBJ2x1, и при этом устройства OBJ1x1 нет в проекте, то название исходного устройства останется без изменений и не будет удалено, иначе название модифицируется по старому правилу в OBJ1x1.
```
PR #1400 (2024-07-17): Fix modify dev obj name
Fixes #1399.

```ChangeLog
Изменения в модификации названий устройств: если при изменении номера объекта (1 -> 2) в объекте было привязаны устройства OBJ2x1, и при этом устройства OBJ1x1 нет в проекте, то название исходного устройства останется без изменений и не будет удалено, иначе название модифицируется по старому правилу в OBJ1x1.
```
closed 2024-07-17 by idzm