sav-oss-eplanner/EasyEPLANner #1408: Не посвечиваются седла микспруфов

KUZMA-1989, 2024-07-29:
Когда курсор ставишь на шаг объекта, то не подсвечиваются седла микспруфов
[image]

KUZMA-1989, 2024-07-30:
@KirillGutyrchik Смотри какой есть нюанс при вставке объектов. В танках прописаны выключения микспруфов и их флипование, когда мы вставляем объект эти клапана остаются неизменными и их нужно менять вручную. Обычно у микспруфов в названии воследняя цифра обозначает номер танка (смотри картинку V113, V3 для разных линий - это означает 3-й танк.). Можно ли при вставке, чтоб в микспруфах привязанных менялась последняя цифра, соответствующая номеру объекта?

KirillGutyrchik, 2024-07-30:
@KUZMA-1989, если у всех микспруфов такая нумерация везде, то да, можно попробовать сделать.

KUZMA-1989, 2024-07-30:
> @KUZMA-1989, если у всех микспруфов такая нумерация везде, то да, можно попробовать сделать.

Да

KirillGutyrchik, 2024-07-30:
> > @KUZMA-1989, если у всех микспруфов такая нумерация везде, то да, можно попробовать сделать.
> 
> Да

Тогда можно это в отдельный issue оформить, и подробнее расписать. И для каких подтипов(только микспруфы или еще какие) это применять

commit ca6681de (2024-08-14): Fix highlight devices for groupable actions (#1409)

Fixes #1408.

```ChangeLog
Исправлено отображение подсветки устройств на схеме для групповых действий;
```
PR #1409 (2024-08-14): Fix highlight devices for groupable actions
Fixes #1408.

```ChangeLog
Исправлено отображение подсветки устройств на схеме для групповых действий;
```
closed 2024-08-14 by idzm