sav-oss-eplanner/EasyEPLANner #1410: Замена названия объекта при копировании и вставке

KUZMA-1989, 2024-07-30:
При копировании и вставке нового объекта в поле "автоматическое включение мойки" не поменялось название из объекта GRUNWALD1->GRUNWALD2 
[image]

KirillGutyrchik, 2024-07-30:
Если микспруф привязывается к агрегату, то номера меняются по аналогии?

KUZMA-1989, 2024-07-30:
Лучше делай только для типов **V_DO1_DI2, V_DO2_DI2, V_DO2_DI2_BISTABLE, V_MIXPROOF, V_BOTTOM_MIXPROOF, V_IOLINK_MIXPROOF, V_AS_MIXPROOF, V_AS_DO1_DI2, V_IOLINK_DO1_DI2, V_IOL_TERMINAL_MIXPROOF_DO3**. И только для объектов из группы **Танк**

KUZMA-1989, 2025-03-12:
@KirillGutyrchik 
В танках прописаны выключения микспруфов и их флипование из объектов линий , когда мы вставляем объект эти клапана остаются неизменными и их нужно менять вручную, потому что их названия не соответствуют вставляемому объекту. Обычно у микспруфов в названии последняя цифра обозначает номер танка (смотри картинку V113, V3 для разных линий - это означает 3-й танк.). Нужно при вставке, чтоб в микспруфах привязанных менялась последняя цифра, соответствующая номеру объекта. Делай для всех типов клапанов, если название устройства не соответствует названию объекта (например ОУ "TANK"), куда вставляем, то заменить последнюю цифру на цифру номера объекта куда вставляем, если данного устройства нету, то удалить его из описания

Что-то не работает 2 пункт. Вставлял объект с номером 1 и такой же объект с номером 2, микспруфы не поменяли в конце цифру 

[image]

commit 73edef10 (2024-08-14): Unification of modify device name (#1411)

Fixes #1410.

```ChangeLog
Добавлена модификация названий устройств в соответсвие объекту для доп.свойств операции; 
```

```ChangeLog
Добавлена специальная функциональность для модификации клапанов некоторых типов, привязаных к действиям аппарата: модифицируется часть номера устройства соответсвующая номеру объекта;
```
PR #1411 (2024-08-14): Unification of modify device name
Fixes #1410.

```ChangeLog
Добавлена модификация названий устройств в соответсвие объекту для доп.свойств операции; 
```

```ChangeLog
Добавлена специальная функциональность для модификации клапанов некоторых типов, привязаных к действиям аппарата: модифицируется часть номера устройства соответсвующая номеру объекта;
```
PR #1592 (2025-05-23): [TechObject] Fix modify devices after paste and replace in actions
Fixes #1410.

```ChangeLog
Исправлена модификация названий устройств при копировании и вставке/замене отдельных действий;
```
commit 7fd41e6f (2025-05-23): [TechObject] Fix modify devices after paste and replace in actions (#1592)

Fixes #1410.

```ChangeLog
Исправлена модификация названий устройств при копировании и вставке/замене отдельных действий;
```
closed 2025-05-23 by KirillGutyrchik