sav-oss-eplanner/EasyEPLANner #1412: Трансляция параметр -> AO

KUZMA-1989, 2024-08-01:
Сделать возможность трансляции **параметр -> AO**, чтоб можно было транслировать параметр из объекта в сигнал AO
[image]

KirillGutyrchik, 2024-08-01:
Для пояснения: 
- Поле для всего действия или для каждой группы по отдельности? 
- В поле прописывается сигнал или float-параметр из объекта, и что оно должно делать?

KUZMA-1989, 2024-08-01:
> Для пояснения:
> 
> * Поле для всего действия или для каждой группы по отдельности?
> * В поле прописывается сигнал или float-параметр из объекта, и что оно должно делать?

Сейчас действующий функционал поля - трансляция сигнала AI в сигнал AO. Я хочу его расширить, чтоб еще работала трансляция значения из параметра Float в сигнал AO

KUZMA-1989, 2024-08-01:
> Для пояснения:
> 
> * Поле для всего действия или для каждой группы по отдельности?
> * В поле прописывается сигнал или float-параметр из объекта, и что оно должно делать?

Не понимаю вопрос из 1 пункта

KirillGutyrchik, 2024-08-01:
> > Для пояснения:
> > 
> > * Поле для всего действия или для каждой группы по отдельности?
> > * В поле прописывается сигнал или float-параметр из объекта, и что оно должно делать?
> 
> Не понимаю вопрос из 1 пункта

Чтобы параметр был в каждой группе:
 - `Группы AI -> AO AO ..`
   + `Группа`
      - `Сигналы`
      - `Параметр`
   + `Группа`
      - `Сигналы`
      - `Параметр`
         . . . 

Или один для всех групп:
 - `Группы AI -> AO AO ..`
   + `Группа` (Сигналы)
   + `Группа` (Сигналы)
       . . .
   + `Параметр`

KirillGutyrchik, 2024-08-01:
@KUZMA-1989, исходя из того, что я понял, могу предложить такой вариант:

 - `Группы AI -> AO AO ..`
   + `Группа`
      - `Источник` (Параметр: указывается либо сигнал `AI` либо float-параметр)
      - `Сигналы` (Указываются сигналы `AO` для трансляции)
   + `Группа`
      - `Источник`
      - `Сигналы`
         . . .

KUZMA-1989, 2024-08-01:
> @KUZMA-1989, исходя из того, что я понял, могу предложить такой вариант:
>
> * `Группы AI -> AO AO ..`
>
>   * `Группа`
>
>     * `Источник` (Параметр: указывается либо сигнал `AI` либо [...]

Типа не как сейчас выглядит поле, что через пробел привязываются сигналы. А будет именно 2 поля?

KirillGutyrchik, 2024-08-01:
> Типа не как сейчас выглядит поле, что через пробел привязываются сигналы. А будет именно 2 поля?

да

KUZMA-1989, 2024-08-01:
> да

А что будет со старыми проектами, где уже указаны данные пары?

KirillGutyrchik, 2024-08-01:
> А что будет со старыми проектами, где уже указаны данные пары?

Ну тут еще нужно подумать, на новую версию можно будет легко переводить, а вот в обратную сторону не получится

KUZMA-1989, 2024-08-01:
> Ну тут еще нужно подумать, на новую версию можно будет легко переводить, а вот в обратную сторону не получится

А нельзя оставить как есть в отображении, только чтоб сигнал поменялся на параметр. Чтоб отображался на 1 позиции либо сигнал, либо параметр

KirillGutyrchik, 2024-08-01:
> А нельзя оставить как есть в отображении, только чтоб сигнал поменялся на параметр. Чтоб отображался на 1 позиции либо сигнал, либо параметр

Можно попробовать сделать, чтобы визуально в редакторе было разделение как я говорил на 2 поля, но сохранять в одну строку как было. Тогда, если будет привязан сигнал, то переходить между версиями можно будет без вреда, но если будет привязан параметр, то уже могут возникнуть проблемы.

KirillGutyrchik, 2024-08-01:
> Можно попробовать сделать, чтобы визуально в редакторе было разделение как я говорил на 2 поля, но сохранять в файле в одну строку как было. Тогда, если будет привязан сигнал, то переходить между версиями можно будет без вреда, но если будет привязан параметр, то уже могут возникнуть проблемы.

На старой версии в таком случае не прочитаются параметры и в поле остануться только сигналы `AO`;
Даже если сохранять в двух форматах (и новом и старом), то опять же в случае с сигналами, все будет хорошо, а вот привязанные параметры в старой версии не прочитаются;

KUZMA-1989, 2024-08-01:
@KirillGutyrchik Главное, чтоб старое открылось в новых полях. Те сигналы, которые были уже прописаны отобразились в новом интерфейсе (если ты будешь делать через 2 поля). А новое - это новое, его уже будут добавлять после обновления

KUZMA-1989, 2024-08-01:
@KirillGutyrchik Я так понимаю Диме надо функционал дописывать новый, чтоб с параметром работал функционал в управляющей программе?

KirillGutyrchik, 2024-08-01:
> @KirillGutyrchik Я так понимаю Диме надо функционал дописывать новый, чтоб с параметром работал функционал в управляющей программе?

Да, и с ним тоже нужно обсудить итоговый формат

KUZMA-1989, 2024-08-01:
> @KirillGutyrchik Я так понимаю Диме надо функционал дописывать новый, чтоб с параметром работал функционал в управляющей программе?

В старой версии нет параметров, только сигналы

KUZMA-1989, 2024-08-05:
@KirillGutyrchik Данный функционал есть в разделе устройства. Закрываем запрос
[image]

closed 2024-08-05 by KUZMA-1989