sav-oss-eplanner/EasyEPLANner #1416: Добавить ошибку привязки бачка ледводы к танкам

KUZMA-1989, 2024-08-07:
Если бачок лед воды привязан к аппарату "Танк", то в логах вывести ошибку о неправильности привязки бачка ледводы

KirillGutyrchik, 2024-08-07:
@KUZMA-1989, к какому базовому объекту он относится? Я что-то не нахожу.

KUZMA-1989, 2024-08-07:
> @KUZMA-1989, к какому базовому объекту он относится? Я что-то не нахожу.

Бачок откачки лелводы

KirillGutyrchik, 2024-08-07:
> Бачок откачки лёдводы

Все, я нашел. Подумал, что он отмечен как устаревший. Только для него ошибку добавлять?

KUZMA-1989, 2024-08-07:
> > Бачок откачки лёдводы
> 
> Все, я нашел. Подумал, что он отмечен как устаревший. Только для него ошибку добавлять?
Да

KUZMA-1989, 2024-08-07:
> > > Бачок откачки лёдводы
> > 
> > 
> > Все, я нашел. Подумал, что он отмечен как устаревший. Только для него ошибку добавлять?
> > Да

еще "Бойлер" добавь

PR #1418 (2024-08-14): Add error message deny binding to unit
Fixes #1416.

```ChangeLog
Добавлено сообщение об ошибке при привязке Бочка ледводы илл Бойлера к Танку;
```
commit 415a887e (2024-08-14): Add error message deny binding to unit (#1418)

Fixes #1416.

```ChangeLog
Добавлено сообщение об ошибке при привязке Бочка ледводы илл Бойлера к Танку;
```
closed 2024-08-14 by idzm