sav-oss-eplanner/EasyEPLANner #1428: Неправильная генерация переменной с устройствами подачи воды

SergeiSilakov, 2024-09-17:
При генерации оборудования объекта поле Клапан(-а) подачи воды генерируется не верно.
[image]
[image]
Необходимо изменить название поля с **water_v** на **water_V** и добавить генерацию переменной как таблицы даже при одном заданном устройстве.
Как должно быть
[image]

KirillGutyrchik, 2025-10-31:
@SergeiSilakov @AlexandrBehunkov так делать для всех полей устройств? или для определенных???

AlexandrBehunkov, 2025-10-31:
@KirillGutyrchik  Такой формат описания используется только для клапа/ов подачи воды у объекта **Линия выдачи** и датчиков температуры объекта **Танк**