sav-oss-eplanner/EasyEPLANner #1482: Сброс параметров и номера следующего шага в объектах при копировании параметров

KUZMA-1989, 2024-12-11:
Сбрасываются привязанные параметры и номера следующего шага в объектах при копировании параметров из одного объекта в другой (копировал из танка в танк)
[image]

KirillGutyrchik, 2024-12-23:
@KUZMA-1989, это изначально так и задумано, что ривязка параметров будет сбрасываться, если их заменять. Могу предложить 2 варинта:
 - Если lua-имя заменненого параметра совпадает, то его привязку не сбрасывать;
 - Вывод диалогового окна с выбором сбрасывать/не сбрасывать привязку;

KUZMA-1989, 2024-12-23:
> @KUZMA-1989, это изначально так и задумано, что привязка параметров будет сбрасываться, если их заменять. Могу предложить 2 варинта:
>  - Если lua-имя заменненого параметра совпадает, то его привязку не сбрасывать;
>  - Вывод диалогового окна с выбором сбрасывать/не сбрасывать привязку;

Первый вариант

commit 4c91bf2c (2024-12-30): Fix parameter binding after replace parameters (#1495)

Fixes #1482.

```ChangeLog
Исправлен сброс привязки параметров к операциям при замене параметров;
```
PR #1495 (2024-12-30): Fix parameter binding after replace parameters
Fixes #1482.

```ChangeLog
Исправлен сброс привязки параметров к операциям при замене параметров;
```
closed 2024-12-30 by idzm