sav-oss-eplanner/EasyEPLANner #1559: Привязка пастеризаторов к группе танков

KUZMA-1989, 2025-03-10:
В объектах линия, линия наполнения, линия выдачи сделать возможным привязывать к группе танков объекты типа "Пастеризатор"

KirillGutyrchik, 2025-03-11:
@KUZMA-1989, @AlexandrBehunkov 

На привязку всех остальных объектов также оставлять запрет?

AlexandrBehunkov, 2025-03-11:
> [@KUZMA-1989](https://github.com/KUZMA-1989), [@AlexandrBehunkov](https://github.com/AlexandrBehunkov)
> 
> На привязку всех остальных объектов также оставлять запрет?

Да, требуется привязка группы танков или пастеризаторов только.

KUZMA-1989, 2025-03-11:
> @KUZMA-1989, @AlexandrBehunkov 
> 
> На привязку всех остальных объектов также оставлять запрет?

Пока да

KirillGutyrchik, 2025-03-11:
> Да, требуется привязка группы танков или пастеризаторов только.

@AlexandrBehunkov @KUZMA-1989 @idzm 

Просто сейчас поля групп объектов ограничивают привязку по типу (Аппарат/Агрегат). А пастеризатор по описанию относится к агрегату. 

Имеет смысл добавить ограничение по конкретным объектам?

```lua
objectGroups =
{
    tanks = { name = "Группа объектов очереди", allowedObjects = { "units" , "pasteurizator" } },
},
```

Или ввести дополнительную типизацию для некоторых агрегатов, которые схожи (пастеризатор, бачок откачки ...) и уже по ней ограничивать привязку?