sav-oss-eplanner/EasyEPLANner #1565: Ошибка при привязке оборудования или параметра

SergeiSilakov, 2025-04-07:
В объекте "бачок откачки лёд воды" при привязке оборудования верхний предел текущего уровня и нижний предел текущего уровня, можно привязывать устройство (аналоговый сигнал) или параметр, что предусмотрено в описании объекта.
Сейчас при привязке параметра выдаётся сообщение с предупреждением о неправильной привязке.

[image]
Нужно добавить проверку на параметр, если привязан существующий параметр не выводить сообщение!

KirillGutyrchik, 2025-05-19:
@SergeiSilakov 

Я так понимаю, ориентироваться на постфик `_SET_VALUE`?
(Если в описании есть в формате *_SET_VALUE, значит можно указать параметр)

```Lua
    UPPER_SET_VALUE = { name = "Верхний предел текущего уровня" },
    LOWER_SET_VALUE = { name = "Нижний предел текущего уровня" },
```

SergeiSilakov, 2025-05-19:
TOTAL_VOLUME_AI
```
FILL = {
                    name = "Наполнение",
                    params =
                    {
                        active =
                        {
                            TOTAL_VOLUME_AI =
                            {
                                name = "Объем линии",
                                displayObjects = { "signals" },
                            },
```

KirillGutyrchik, 2025-05-19:
Так это параметры, а в запросе речь про Оборудование.

SergeiSilakov, 2025-05-20:
Я не могу подсказать такой пример. @AlexandrBehunkov есть похожий пример в текущем описании?

SergeiSilakov, 2025-05-20:
Возможно такое описание только в бачке появилось

AlexandrBehunkov, 2025-05-20:
@SergeiSilakov @KirillGutyrchik. Данное поле в оборудовании может выступать как значение параметра ( number type), так и аналоговое значение ( AI signal ). На данный момент использовался только параметр.

SergeiSilakov, 2025-05-20:
@KirillGutyrchik Сделай как для поля производительность в "устройства" можно и сигнал и параметр