sav-oss-eplanner/EasyEPLANner #1572: Генерация параметров объектов

KUZMA-1989, 2025-04-21:
- [ ] При добавлении узлов охлаждения, нагревания и т.п., то параметры в новый объект генерируются автоматически. Если объект действующий, то параметры новые не добавляем.
- [ ] При добавлении новых объектов типа **Танк, Линия** параметры добавляются после описания операций `путем нажатия кнопки` **добавить параметры** на поле **"Параметры"**.

KirillGutyrchik, 2025-04-22:
@KUZMA-1989 @SergeiSilakov 

В каком порядке добавлять параметры?

1. Параметры, относящиеся к операциям -> Параметры, относящиеся к агрегатам -> +2 заглушки
2. Параметры, относящиеся к агрегатам -> Параметры, относящиеся к операциям -> +2 заглушки

Или другой какой-то порядок?

KUZMA-1989, 2025-04-22:
> [@KUZMA-1989](https://github.com/KUZMA-1989) [@SergeiSilakov](https://github.com/SergeiSilakov)
>
> В каком порядке добавлять параметры?
>
> 1. Параметры, относящиеся к операциям -> Параметры, относящиеся [...]

Сделать Системные параметры (1-20), Мойка (21-30) и далее с диапазоном 10 для всех остальных операций, чтоб генерировалось в зависимости от порядкового номера операции в объекте
Вот пример объектов

KUZMA-1989, 2025-04-22:
[Линия выдачи.txt](https://github.com/user-attachments/files/19845835/default.txt)
[Линия наполнения.txt](https://github.com/user-attachments/files/19845833/default.txt)
[Танк.txt](https://github.com/user-attachments/files/19845834/default.txt)

KirillGutyrchik, 2025-04-22:
@KUZMA-1989, А системные откуда брать? или просто 20 заглушек делать?

KUZMA-1989, 2025-04-22:
> [@KUZMA-1989](https://github.com/KUZMA-1989), А системные откуда брать? или просто 20 заглушек делать?

Я тебе для чего скинул файлы экспорта? Там все есть

KirillGutyrchik, 2025-04-22:
@SergeiSilakov, получается, нужно немножоко переделать описание?

Например:
```Lua
Parameters = 
{
    system_float = {
        -- Общие системные параметры (добавляются в начало и могут быть в нескольких операциях) 
    }
    float = {
        -- Параметры относящиеся к операции
    }
}
```

и как я понимаю `OPERATION_TIME` и `STEP_TIME` есть везде.

KUZMA-1989, 2025-04-22:
> [@SergeiSilakov](https://github.com/SergeiSilakov), получается, нужно немножоко переделать описание?
> 
> Например:
> 
> parameters = 
> {
>     system_float = {
>         -- Общие системные параметры (добавляются в начало и могут быть в нескольких операциях) 
>     }
>     float = {
>         -- Параметры относящиеся к операции
>     }
> }
> и как я понимаю `OPERATION_TIME` и `STEP_TIME` есть везде?

ДА

KirillGutyrchik, 2025-04-25:
@KUZMA-1989 @SergeiSilakov 

Параметры которые генерируются из доп.свойст привязанных агрегатов подставлять в область параметров, пренадлежащей этой операции?

[image]

KUZMA-1989, 2025-04-25:
> @KUZMA-1989 @SergeiSilakov 
> 
> Параметры которые генерируются из доп.свойст привязанных агрегатов подставлять в область параметров, пренадлежащей этой операции?
> 
> [image]

Может наоборот стрелка должна стоять?

KirillGutyrchik, 2025-04-25:
> Может наоборот стрелка должна стоять? Есть параметры объекта и они прописываются в описание параметров операции

Все так, но я тут другое имею ввиду: эти параметры для агрегатов заполняются автоматически, и вот их подставлять в ту же область параметров, где генерируются параметры для операции  вместо заглушек, или их генерировать отдельно в конце?

SergeiSilakov, 2025-04-25:
Вместе с операциями

SergeiSilakov, 2025-04-25:
Параметры танка (общие)
заглушки...
Параметры операции №1
Параметры для узлов операции №1
Заглушки...
Параметры оп. №2
Параметры узлов операции №2
....

commit 6a7591ff (2025-05-19): [TechObject] Fix autocomplete parameters (#1580)

Fixes #1572.

```ChangeLog
Измения в автозаполнении параметров: автозаполение параметров операций и параметров привязанных узлов по всем операциям, изменение структуры генерации параметров;
```

---------
PR #1580 (2025-05-19): [TechObject] Fix autocomplete parameters
Fixes #1572.

```ChangeLog
Измения в автозаполнении параметров: автозаполение параметров операций и параметров привязанных узлов по всем операциям, изменение структуры генерации параметров;
```
closed 2025-05-19 by idzm