sav-oss-eplanner/EasyEPLANner #1590: Изменение структуры описания

SergeiSilakov, 2025-05-19:
Изменился номер поля в операции "Охлаждение после сквашивания"
Пример на скриншоте:

[image]

SergeiSilakov, 2025-05-20:
[image]

KirillGutyrchik, 2025-05-20:
@SergeiSilakov 

В общем это Lua шалит, это появвилось после добавления нового параметра `OPERATION_AFTER`.
Так как у нас это поле - не массив, то оно читается в неком произвольном порядке, и после добавления нового параметр порядок изменился🙃.

```Lua
active =
{
    COOLING_TIME =
    {
        name = "Время охлаждения (1-го шага)",
        displayObjects = { "parameters" },
    },
    MIXING_CHECK_TIME =
    {
        name = "Время проверки температуры (3-го шага)",
        displayObjects = { "parameters" },
    },
    OPERATION_AFTER = { name = "Номер последующей операции" },
},
```

по хорошему бы изменить описание, чтобы всегда был единый порядок:
```Lua
active =
{
    {
        luaName = "COOLING_TIME"
        name = "Время охлаждения (1-го шага)",
        displayObjects = { "parameters" },
    },
    {
        luaName = "MIXING_CHECK_TIME"
        name = "Время проверки температуры (3-го шага)",
        displayObjects = { "parameters" },
    },
    {
        luaName = "OPERATION_AFTER"
        name = "Номер последующей операции" 
    },
},
```

KirillGutyrchik, 2025-05-20:
Либо можно при чтении сортировать параметры по назвпнию, но тогда, вероятно порядок тоже много где поменяется и скорее всего будет отличаться от того, который записан в базовом описании.

SergeiSilakov, 2025-05-21:
@KirillGutyrchik ок, оставляем выбор за lua

KirillGutyrchik, 2025-05-21:
Закрываем PR?

SergeiSilakov, 2025-05-21:
@KirillGutyrchik ДА

closed 2025-05-21 by KirillGutyrchik