sav-oss-eplanner/EasyEPLANner #1593: Lua-имя в параметрах

KirillGutyrchik, 2025-05-23:
Добавить првоерку Lua-имени для параметра.

KirillGutyrchik, 2025-05-23:
@SergeiSilakov @KUZMA-1989 

Может есть смысл перенести Lua-название во второй столбец для удобства, что скажете?

[image]

KUZMA-1989, 2025-05-23:
> [@SergeiSilakov](https://github.com/SergeiSilakov) [@KUZMA-1989](https://github.com/KUZMA-1989)
> 
> Может есть смысл перенести Lua-название во второй столбец для удобства, что скажете?
> 
> [image]

Напротив названия на кирилице?

KirillGutyrchik, 2025-05-23:
> Напротив названия на кирилице?

Да

KirillGutyrchik, 2025-05-23:
Типа такого

[image]

KUZMA-1989, 2025-05-23:
> Типа такого
> 
> [image]

Делай

PR #1594 (2025-06-04): [TechObject] Refactoring editing luaName in parameters
Fixes #1593.

```ChangeLog
Изменения в редактировании Lua-имени параметров: теперь название задается во второй колонке + проверка названия;
```