sav-oss-eplanner/EasyEPLANner #1595: Описание логики поля задания

SergeiSilakov, 2025-06-03:
Добавить в поля описания Eplanera описание логики для указанных сигналов.
Например:

[image]

KirillGutyrchik, 2025-06-05:
@SergeiSilakov, а в полях DO DI устройств логика 'ИЛИ'?

KirillGutyrchik, 2025-06-05:
То, что на скрине, тут или?
[image]

SergeiSilakov, 2025-06-05:
Переход к состоянию - "И"

SergeiSilakov, 2025-06-05:
Включение текущего шага - "И"

SergeiSilakov, 2025-06-05:
Переход к следующему шагу - "И"

KirillGutyrchik, 2025-06-05:
Не, подожди, все это в документации есть, я спрашивал только про "Устройства", так как там не совсем понятно

KirillGutyrchik, 2025-06-05:
Хмм, а что, если может понадобится активация не одного из сигналов, а всех вместе? и немного не полнимаю, что делает поле DO?