sav-oss-eplanner/EasyEPLANner #1671: Двойная подсветка устройств

KUZMA-1989, 2025-09-28:
Вернуть функционал, когда клапаны прописаны и во **включении** и в **выключении**, то они были подсвечены двумя цветами **наполовину зеленым и серым**. 
Сейчас данный функционал пропал. Было очень удобно видеть, когда по ошибке прописал в 2-х местах

[image]

KirillGutyrchik, 2025-09-29:
@KUZMA-1989, сложно, какие еще есть исключение кроме включение-выключение?

KUZMA-1989, 2025-09-30:
> [@KUZMA-1989](https://github.com/KUZMA-1989), [@SergeiSilakov](https://github.com/SergeiSilakov), [@AndreyNemenkov](https://github.com/AndreyNemenkov), [@painterlv](https://github.com/painterlv)
> 
> Предлагаю, если идет неверная привязка: подсвечивать просто крассным цветом. Только что с двойными клапанами, если они допустим привязаны к выключать и нижние/верхние седла, такое может быть?
> 
> И основные цвета не менять? так и остается зеленый - вкл, сервый - выкл? могу добавить доп. цветов для каких-то специфических действий
Давай красным, если и на включении и на выключении устройство прописано. Остальной функционал не меняй

closed 2025-10-30 by idzm